Steffen-W/KiCad-Parasitics #2: unhandled Fatal Error

nopeppermint, 2023-10-01:
Hi,

Thanks for this Plugin.

I tried it out with Kicad and selected a track and got the following error:

[image]

This project I can not share.

But another fatal error is also available on kicads demo project stickhub:

1. open demo project stickhub
2. open pcb file
3. select the track on the usb d+ line
4. click on the parasitic icon of Kicad-Parasitics addon
5. fatal error

[image]

```
Application: KiCad PCB Editor x86_64 on x86_64

Version: 7.0.8-rc1-202309282119~cd3785744c~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.16

Platform: Linux Mint 21.2, 64 bit, Little endian, wxGTK, cinnamon, x11

Build Info:
	Date: Sep 28 2023 21:19:16
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 11.4.0 with C++ ABI 1016

Build settings:
	KICAD_SPICE=ON

```

nopeppermint, 2023-10-01:
some other Fatal errors:

[image]

nopeppermint, 2023-10-01:
some more:

[image]

[test_project.zip](https://github.com/Steffen-W/KiCad-Parasitics/files/12776989/test_project.zip)

Steffen-W, 2023-10-01:
Hi @nopeppermint, thanks for that issue. I am a bit puzzled about the messages right now and will try to find out what went wrong. Your error messages look different. At first I thought you were using kicad 6 which is definitely not the case. That in one case the reference to the pad is not found is a harmless but also strange error. Nothing is done with the value in the following. I have planned something else with it but it is not implemented.

 The message with the cu stack indicates to me that my implementation is too unstable. I ask to excuse this. If you could send me the first lines from the ".kicad_pcb" it would be very helpful. But only the ones up to the layer structure.

I would like to ask you nevertheless with the whole errors whether the Plugin with you at all functioned or exclusively error produces? 

Thanks for your message.

nopeppermint, 2023-10-01:
[stickhub_copy.zip](https://github.com/Steffen-W/KiCad-Parasitics/files/12777514/stickhub_copy.zip)

I was never able to use the plugin, I only got errors.

Maybe you could share on the project repository also an example kicad project which works on your side?

Steffen-W, 2023-10-01:
Unfortunately, I am currently on the road and do not have my PC available. Maybe I still manage to locate the errors. 

You have also used the latest version of the plugin which I assume (2023.09.29)?

My setup with you corresponds to what I have tested. The only thing that can be is that it is due to the created projects and the settings contained therein because of which it does not work. I hope actually that I get along with it but that is not so here. I see that I set up kicad on a PC and test it.

nopeppermint, 2023-10-01:
>Unfortunately, I am currently on the road and do not have my PC available. Maybe I still manage to locate the errors.

no worries, I was just curious what the plugin does, and as it didn't worked for me I started posting error reports.
Feel free to work on it whenever you find time, no hurry.

> You have also used the latest version of the plugin which I assume (2023.09.29)?

yes I am using 2023.09.29

Steffen-W, 2023-10-01:
No problem I want myself that you can do something with the plugin. If the plugin does not work right away, that is a deterrent for most people. Currently, the plugin can determine a parasitic DC resistance as well as calculate an inductance. Unfortunately, the inductance calculation only assumes constructive superposition of the fields and possible destructive not. 

To your error messages. 
It seems to me that in test_project.kicad_pcb no layer structure is defined. This is a problem for the plugin, because it does not know how thick the copper layer is and also cannot calculate the length of the vias. An understandable error message would be appropriate here.

In your other project it was a different problem. I would just comment out the corresponding line. This has no further effect. The associated footprint is currently ignored.

I will correct the problems but it will take some time.

Steffen-W, 2023-10-10:
Hi @nopeppermint ,

can you test the latest release to let me know if the plugin works for you everywhere now? I would be very grateful.
[Release 2023.10.10](https://github.com/Steffen-W/KiCad-Parasitics/releases/tag/2023.10.10)

nopeppermint, 2023-10-10:
yes, that one works!

I manually installed it, as the kicad plugin manager package seems not yet to be updated.

I tested some distance on the mentioned Kicad Demo Project Stickhub.

I verified that the calculated resistance change when I change the track width between the two selected points.

Apparently changing the copper thickness in the PCB stackup has no influence of the resistor calculation.

Would be nice if you could show more details in the result window, what is actually calculated and based on which parameters.

Steffen-W, 2023-10-11:
Hi @nopeppermint,

I think it's great that you took a closer look at the code. I have just incorporated your comments and hope I have understood you correctly. I will continue to develop the plugin. I would also like the parasitic capacity to be displayed soon. With the formula for the inductance I am unfortunately still dissatisfied. The results are unfortunately only rough orientations and not usable for pcb coil, which was my goal. The resistance is currently calculated quite well in my opinion. Soon the skin effect for higher frequencies will be considered. Polygone will be a perfect conductor in the future, because without FEM simulation I have no chance.

I am happy about any comments and am open-minded if you think of improvements or functional extensions.

nopeppermint, 2023-10-11:
thanks for the fix

closed 2023-10-11 by nopeppermint