Steffen-W/KiCad-Parasitics #5: Selection bug

a-allard, 2023-10-13:
Hello.  Thanks for making this plugin!  I had to tweak it a bit to get it running on my system but after I got it running I was really happy with it.

### My system:
Windows 11
#### Kicad info
Application: KiCad x64 on x64

Version: 7.0.8, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Sep 29 2023 18:44:47
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.7.1
	Curl: 7.88.1-DEV
	ngspice: 41
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

#### Mods I made:
I only changed Get_PCB_Elements.py
```python
import numpy as np
import pcbnew

ToMM = pcbnew.ToMM

def SaveDictToFile(dict_name, filename):
    with open(filename, "w") as f:
        f.write("data = {\n")
        for uuid, d in list(dict_name.items()):
            f.write(str(uuid))
            f.write(":")
            f.write(str(d))
            f.write(",\n")
        f.write("}")

# Überprüfe, ob der Punkt sich innerhalb des Polygons befindet
def IsPointInPolygon(point_, polygon_):
    point = np.array(point_)
    polygon = np.array(polygon_)

    n = len(polygon)
    inside = False

    p1x, p1y = polygon[0]
    for i in range(n + 1):
        p2x, p2y = polygon[i % n]
        if point[1] > min(p1y, p2y):
            if point[1] <= max(p1y, p2y):
                if point[0] <= max(p1x, p2x):
                    if p1y != p2y:
                        x_intersect = (point[1] - p1y) * (p2x - p1x) / (p2y - p1y) + p1x
                    if p1x == p2x or point[0] <= x_intersect:
                        inside = not inside
        p1x, p1y = p2x, p2y

    return inside

[... 187 lines cut ...]
            continue

        temp["Netname"] = Pad.GetNetname()
        temp["NetCode"] = Pad.GetNetCode()
        temp["id"] = getHash(Pad)
        temp["IsSelected"] = Pad.IsSelected()
        temp["connStart"] = sorted(
            getHashList(connect.GetConnectedPads(Pad))
            + getHashList(connect.GetConnectedTracks(Pad))
        )
        ItemList[temp["id"]] = temp

    for uuid, d in list(ItemList.items()):  # TODO: WIRES still need to be considered
        if d["type"] == "ZONE":
            for item in d["connStart"]:
                if not "connEND" in ItemList[item]:
                    ItemList[item]["connStart"].append(uuid)

    return ItemList

```

If you want I can fork the project and create a PR.

Thanks again for the plugin

Steffen-W, 2023-10-13:
Hi @a-allard,

Thank you for your efforts. I am just trying to understand what the problem could be.
Is the type() function not working for you? You have converted the variable to a string and compared it to a string. In my opinion the previous variant should also work. What is the error message?

Maybe it helps to set the debug variable. [__init__.py#L21](https://github.com/Steffen-W/KiCad-Parasitics/blob/c43bbeff3a41bcb0973db4e5f42c63661b4dba59/plugins/__init__.py#L21)

I have a feeling that you have a problem with the pcbnew.ToMM function.

``` python
def ToMM(iu):
    if type(iu) in [int,float]:
        return float(iu) / float(IU_PER_MM)
    elif type(iu) in [wxPoint,wxSize,VECTOR2I]:
        return tuple(map(ToMM,iu))
    else:
        raise TypeError("ToMM() excpects int, float, wxPoint, wxSize or VECTOR2I, instead got type " + str(type(iu)))
```
Can you find out what type of this function is passed in your places?
Would be very happy if we find the error together.

a-allard, 2023-10-23:
Hi @Steffen-W 
Sorry for the delay...

The issue seems 2 fold.  The first is a weird bug with the type function call.  I don't know why it doesn't work.  

I opened up the scripting terminal as it was easy to show.

When trying to use the type function it doesn't seem to work with pcbnew types...

I grabbed the all the tracks into a list.  Then to double check the first element was a track: ```type(tcs[0])```
That returns:
><class 'pcbnew.PCB_TRACK'>

But when trying to see if the type is a track with: ```type(tcs[0]) is pcbnew.PCB_TRACK``` returns:
>False

So the script the way it works doesn't think there are any tracks, pads, etc.  But when I use: ```'pcbnew.PCB_TRACK' in str(tcs[0])``` I get:
>True

The second has to do with the ToMM built in function.  I ended up converting the call `ToMM()` to `tuple(map(ToMM, iu))` I got an error about a VECTOR2I not working in the function even though it is supposed to be one of the accepted types...

As an example:

```
bd = pcbnew.GetBoard()
tcs = bd.GetTracks()
tcs[0].GetStart()
```

returns:

>VECTOR2I(150187744, 120400000)

A VECTOR2I should should be just fine to drop into the ToMM function.  When I try to do so:

```
pcbnew.ToMM(tcs[0].GetStart())
```

I get:

>Traceback (most recent call last):
  File "/usr/lib/python3.10/code.py", line 90, in runcode
    exec(code, self.locals)
  File "<input>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 19272, in ToMM
    raise TypeError("ToMM() expects int, float, wxPoint, wxSize or VECTOR2I, instead got type " + str(type(iu)))
TypeError: ToMM() expects int, float, wxPoint, wxSize or VECTOR2I, instead got type <class 'pcbnew.VECTOR2I'>

Steffen-W, 2023-10-24:
Hi @a-allard ,

To be honest, I am a bit confused as to what the problem is. It will have something to do with "import pcbnew".
I just typed the following into KiPython (Scriptin Console). Do you get an error with the lines?
``` python
>>> import pcbnew
>>> bd = pcbnew.GetBoard()
>>> Tracks = bd.GetTracks()
>>> Track = Tracks[0]
>>> Track.GetStart()
VECTOR2I(160950000, 44200000)
>>> pcbnew.ToMM(Track.GetStart())
(160.95, 44.2)
>>> type(Track) is pcbnew.PCB_TRACK
True
>>> 'pcbnew.PCB_TRACK' in str(Track)
True
```

I have unfortunately just yet no concrete idea where the error lies and would try to find out little by little about testing. So far, I can not reproduce the error.

a-allard, 2023-10-24:
That is very strange....  If I run the exact same code you have there I get:

```python
>>> import pcbnew
>>> bd = pcbnew.GetBoard()
>>> Tracks = bd.GetTracks()
>>> Track = Tracks[67] # 0 isn't a track on this board I opened this morning
>>> Track.GetStart()
VECTOR2I(160950000, 44200000)
>>> pcbnew.ToMM(Track.GetStart())  # The error generated here make no sense as the code definitely should support this...
Traceback (most recent call last):
  File "/usr/lib/python3.10/code.py", line 90, in runcode
    exec(code, self.locals)
  File "<input>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 19272, in ToMM
    raise TypeError("ToMM() expects int, float, wxPoint, wxSize or VECTOR2I, instead got type " + str(type(iu)))
TypeError: ToMM() expects int, float, wxPoint, wxSize or VECTOR2I, instead got type <class 'pcbnew.VECTOR2I'>
>>> tuple(map(pcbnew.ToMM, Track.GetStart())) # but this works...
(141.2621, 103.733193)
>>> type(Track) is pcbnew.PCB_TRACK
False
>>> 'pcbnew.PCB_TRACK' in str(Track)
True

```

Also this is happening to me on Ubuntu 22.04 and Windows 11.  Are you running on MacOs?

Steffen-W, 2023-10-24:
Hi @a-allard , 

Just tested it on OS: Linux Mint 21.2 x86_64.
But I can also test it under windows. Your error is really funny. Is the installation corrupted somehow? Can you upload me the pcbnew.py file? (/usr/lib/python3/dist-packages/pcbnew.py)

a-allard, 2023-10-24:
@Steffen-W 

Interesting.  I agree the error is really weird and doesn't make sense.  Attached should be the file.  I had to make it a txt as github was being dumb.

[pcbnew.txt](https://github.com/Steffen-W/KiCad-Parasitics/files/13113112/pcbnew.txt)

Steffen-W, 2023-10-24:
Hi @a-allard ,

Your pcbnew.py
``` python
def ToMM(iu):
    if type(iu) in [int,float]:
        return float(iu) / float(pcbIUScale.IU_PER_MM)
    elif type(iu) in [wxPoint,wxSize,VECTOR2I]:
        return tuple(map(ToMM,iu))
    else:
        raise TypeError("ToMM() expects int, float, wxPoint, wxSize or VECTOR2I, instead got type " + str(type(iu)))
```

My version
``` python
def ToMM(iu):
    if type(iu) in [int,float]:
        return float(iu) / float(IU_PER_MM)
    elif type(iu) in [wxPoint,wxSize,VECTOR2I]:
        return tuple(map(ToMM,iu))
    else:
        raise TypeError("ToMM() excpects int, float, wxPoint, wxSize or VECTOR2I, instead got type " + str(type(iu)))
```

[pcbnew.py](https://github.com/Steffen-W/KiCad-Parasitics/files/13113202/pcbnew_py.txt)

I'm stumped right now as to why your verion looks so different. Do you still have a KiCad 6 installation and your file is derived from it?

a-allard, 2023-10-24:
Hi @Steffen-W 

Well that is in very strange.  The pcbnew files look identical.

I just searched the entire `/usr/lib` area and only turned up one pcbnew.py file.  The one I just shared.

Here is the version of Kicad from Ubuntu:

Application: KiCad PCB Editor x86_64 on x86_64

Version: 7.0.8-7.0.8~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.16

Platform: Ubuntu 22.04.3 LTS, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Sep 30 2023 13:54:18
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 11.4.0 with C++ ABI 1016

Build settings:
	KICAD_SPICE=ON

Installed from this ppa:

https://ppa.launchpadcontent.net/kicad/kicad-7.0-releases/ubuntu/ (jammy main)

On the windows machine I have it just downloaded from the kicad.org site.

I don't know either.  I should not be getting the errors I get.  There is something wrong with the type function but I don't know what it is...

a-allard, 2023-10-24:
One other thing....

On the linux machine for the scripting console, I have py 0.9.8 running in python 3.10.12 gcc 11.4.0

Steffen-W, 2023-10-24:
My just tested setup is:

```
Application: KiCad x86_64 on x86_64
Version: 7.0.8, release build
Libraries:
    wxWidgets 3.2.3
    FreeType 2.13.2
    HarfBuzz 8.1.1
    FontConfig 2.14.2
    libcurl/8.4.0-DEV OpenSSL/3.1.3 zlib/1.3 nghttp2/1.56.0
Platform: Freedesktop.org SDK 23.08 (Flatpak runtime), 64 bit, Little endian, wxGTK, cinnamon, x11
Build Info:
    Date: Oct 21 2023 18:28:07
    wxWidgets: 3.2.3 (wchar_t,wx containers) GTK+ 3.24
    Boost: 1.83.0
    OCC: 7.7.2
    Curl: 8.4.0-DEV
    ngspice: 41
    Compiler: GCC 13.2.0 with C++ ABI 1018
Build settings:
    KICAD_SPICE=ON
```

Steffen-W, 2023-10-24:
The version in the documentation is also the one from me. I am trying to find out where your version comes from.

https://docs.kicad.org/doxygen-python-6.0/pcbnew_8py_source.html#l18131

a-allard, 2023-10-24:
Very odd.

I checked the pcbnew on my windows machine.  With the exception of a few functions defined in the ubuntu version that are missing from the windows one the windows and ubuntu file seem to match up.  (ie any difference is just a function in the ubuntu py file that is not in the ubuntu one) 

Attached for reference.
[pcbnew_windows_py.txt](https://github.com/Steffen-W/KiCad-Parasitics/files/13113726/pcbnew_windows_py.txt)

Steffen-W, 2023-10-24:
Hi @a-allard,  the part relevant to this problem is actually the same for you in linux and windows. But the files are different. For example there is BOARD_GetStandardLayerName in linux which does not exist in windows. Unfortunately, this does not give me any information about what is wrong. Is there the same error in Windows?

a-allard, 2023-10-24:
Hi @Steffen-W 

Yes the errors are the same.
```python
>>> import pcbnew
>>> bd = pcbnew.GetBoard()
>>> tracks = bd.GetTracks()
>>> tracks[0]
<pcbnew.PCB_TRACK; proxy of <Swig Object of type 'PCB_TRACK *' at 0x00000220A5EF9570> >
>>> track = tracks[0]
>>> pcbnew.ToMM(track.GetStart())
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\pcbnew.py", line 18690, in ToMM
    raise TypeError("ToMM() expects int, float, wxPoint, wxSize or VECTOR2I, instead got type " + str(type(iu)))
TypeError: ToMM() expects int, float, wxPoint, wxSize or VECTOR2I, instead got type <class 'pcbnew.VECTOR2I'>
>>> type(track) is pcbnew.PCB_TRACK
False
```

Steffen-W, 2023-10-24:
Can you test this once @a-allard ?

``` Python
>>> import pcbnew
>>> from pcbnew import VECTOR2I
>>> bd = pcbnew.GetBoard()
>>> Tracks = bd.GetTracks()
>>> Track = Tracks[0]
>>> isinstance(Track.GetStart(), VECTOR2I)
True
```

a-allard, 2023-10-24:
Sure.  I can test on windows.  For ubuntu it'll be tonight.

Windows

```python
>>> isinstance(track.GetStart(), pcbnew.VECTOR2I)
False
```

a-allard, 2023-10-24:
Something with type is really broken.  But the normal python types seem to work okay

```python
>>> type(123) is int
True
>>> type([]) is list
True
>>> type(0.0) is float
True
```

I have no idea what would cause type to act weird in pcbnew types but fine for regular python types...

Steffen-W, 2023-10-24:
It really sucks that I haven't found a way to create the same error yet. I don't have the error on Windows either.

just tested
```
Application: KiCad x64 on x64
Version: 7.0.5, release build
Libraries:
    wxWidgets 3.2.2
    FreeType 2.12.1
    HarfBuzz 6.0.0
    FontConfig 2.14.1
    libcurl/7.88.1-DEV Schannel zlib/1.2.13
Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Date: May 27 2023 02:48:13
    wxWidgets: 3.2.2 (wchar_t,wx containers)
    Boost: 1.81.0
    OCC: 7.6.3
    Curl: 7.88.1-DEV
    ngspice: 40
    Compiler: Visual C++ 1936 without C++ ABI
Build settings:
    KICAD_SPICE=ON
```
I'll try it right away with a mac yet

Steffen-W, 2023-10-24:
Does something so trivial work?

```
>>> pcbnew.VECTOR2I_MM(1,2)
VECTOR2I(1000000, 2000000)
>>> isinstance(pcbnew.VECTOR2I_MM(1,2), pcbnew.VECTOR2I)
True
```

a-allard, 2023-10-24:
I agree.  Really weird...

The trivial things work.

```python
>>> pcbnew.VECTOR2I_MM(1,2)
VECTOR2I(1000000, 2000000)
>>> isinstance(pcbnew.VECTOR2I_MM(1,2), pcbnew.VECTOR2I)
True
```

a-allard, 2023-10-24:
Okay....  Some more weird

If I force the VECTOR2I point returned from `track.GetStart()` to be a VECTOR2I 'ToMM' works. As does type.

```python
>>> pcbnew.VECTOR2I(track.GetStart())
VECTOR2I(113620000, 91595000)
>>> pcbnew.ToMM(pcbnew.VECTOR2I(track.GetStart()))
(113.62, 91.595)
>>> pcbnew.ToMM(track.GetStart())
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\pcbnew.py", line 18690, in ToMM
    raise TypeError("ToMM() expects int, float, wxPoint, wxSize or VECTOR2I, instead got type " + str(type(iu)))
TypeError: ToMM() expects int, float, wxPoint, wxSize or VECTOR2I, instead got type <class 'pcbnew.VECTOR2I'>
>>> isinstance(pcbnew.VECTOR2I(track.GetStart()), VECTOR2I)
True
>>> type(pcbnew.VECTOR2I(track.GetStart()))is VECTOR2I
True

```

Is KiCad making a different version of VECTOR2I when calling `pcbnew.GetBoard().GetTracks()[0].GetStart()`?  Just seems like there is some very slight difference and python won't recognize them as the same thing...

Steffen-W, 2023-10-24:
I find the error really strange. Make me soon times on the search. Unfortunately, I do not have the time now to understand exactly what is wrong. But it will be something with the KiCad Python library. Maybe you are right with your assumption but that would surprise me very much (I currently have no better explanation).

https://gitlab.com/kicad/code/kicad/-/issues

I almost don't think I can avoid writing an issue with the little test that doesn't work for you. @a-allard if you like you are welcome to do that, would be very happy about it.

a-allard, 2023-10-24:
Hi @Steffen-W 

Fair enough.  I can open an issue in the kicad repo later today with all the details.

If you're good with it I'll leave this issue open for the time being.  At least until we have confirmation from the kicad team it is one their end.

Thanks for the help working on this!

Cheers.

Steffen-W, 2023-10-24:
Hi @a-allard,

I also thank you for your efforts. It is really very helpful when you get support when looking for errors. I would like the plugin to work for everyone, even if it is still a long way to go until I am satisfied with the plugin. I will not close the issue until the problem is fixed.
Please write the link to the issue when you open one.

a-allard, 2023-10-24:
Sounds like a plan!

[link to kicad issue](https://gitlab.com/kicad/code/kicad/-/issues/15943)

a-allard, 2023-10-25:
Hi @Steffen-W 

So......  The kicad developers mentioned other plugins.  After some experimentation I noted KiKit was causing the issue.  After mentioning this to the Kicad guys the linked me over to [this issue](https://github.com/yaqwsx/pcbnewTransition/issues/3)

This is turning into quite a goose chase!  Anyway.  Without KiKit your plugin works perfectly without any mods.  Closing this as it is something in a completely different plugin.

Thanks again for the help!

closed 2023-10-25 by a-allard