Steffen-W/KiCad-Parasitics #6: ERROR in Resistance Network calculation (ngspice)

Bassude, 2023-10-25:
Dear maintainer,

when calculating a connection, the results are displayed (--> Resistance (only short path) = xx.xxx mOhm) but followed by an error message, indicating that no ngspice installation could be found. As you can see below, ngspice is configured and available.

**parasitic version: 2023.10.11	22 kB	67 kB	✔	stable**

Application: KiCad PCB Editor x64 on x64

Version: 7.0.1, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 5.0.1
	FontConfig 2.14.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 11 2023 03:41:46
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.83.1-DEV
	**ngspice: 39**
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_SPICE=ON

Any suggestions?
Is the error message the error itself as ngspice is working?

Regards,

Bassude

Steffen-W, 2023-10-25:
Hi @Bassude ,

I should have fixed the problem already. But I haven't published it as a release yet. But I will do it soon. The code is already available in Git.

Bassude, 2023-10-26:
Thanks!
The issue mentioned above was resolved with the current code (Oct. 26). While installing the package, there seems to be an issue with the metadata now.

Anyhow, it still runs as is.

[image]

Steffen-W, 2023-10-27:
Hi @Bassude ,

The script generate_zip.sh creates a zip document with the current version number. I would be happy if you check if the plugin works for you as desired.
https://github.com/Steffen-W/KiCad-Parasitics/releases/download/2023.10.27/KiCad-Parasitics.zip

Bassude, 2023-10-27:
Hello Steffen,

will check the script on my GNU/Linux machine ~~
Thanks for the new release, I will share this!

Grüße!

Steffen-W, 2023-10-28:
Hi @Bassude was your test successful? 

For the Linux version nothing should have changed. The change of the code was only for windows. Under mac runs the ngspice guaranteed not yet but there I also do not know how to access ngspice. 

What do you want to use the plugin for? What would you be interested in what I integrate into the next version?

Bassude, 2023-10-29:
Hi!

The test was successful, the functionality is there - no error messages. Thanks!

I generally use it to balance parallel connections of power transistors (gate connection).
I believe it can also be useful to optimize oscillator circuits for an MCU.

Features:
* Capacitance to GND plane is already on your radar
* Hotkey to activate the plug-in
* Highlighting the connection (traces) in between the two points and not just the two points itself

I can also check with some colleagues to understand what they would expect.
This will take a while.

Anyhow, thanks for reaching out and the effort you have put into this!

Bassude

Steffen-W, 2023-10-29:
Hi @Bassude, you are always welcome to create a feature request for your issues. I will take care of the capacity. With the displayed resistance values you should consider that they are dc values and not ac values. I will add that soon.

closed 2023-10-29 by Steffen-W