Steffen-W/KiCad-Parasitics #8: Fatal Error with little description

LeanderGlanda, 2024-01-05:
Hi there, I recently installed your plugin as I thought it would be nice to be able to calculate/view properties of my tracks on the PCB I'm currently working on. Today I wanted to try parasitics, but didn't get it to work. I have attached an image below that shows the error.

Some additional information:
* My laptop runs Windows 11
* I use KiCAD 7.0.10 (previously 7.0.5, I updated but that didn't make the plugin work)
* My PCB Stackup in KiCAD is custom so it fits the one I use from JLCPCB.
* My PCB design is unfinished and not all nets are connected.

Don't stress with finding a solution. I just thought I could report the error and help you out with development by reporting the bug.
If you tell me what to do I can help you debug for a solution or provide further information.

[image]

Steffen-W, 2024-01-10:
Hi,

I had seen your issue promptly but did not reply directly because of what was written. I suspect that the problem is actually that your layout is not ready yet. The plugin tries to analyze all lines, no matter if they are necessary or not. When you have finished your layout, I would be happy if you could give me an update.

I will solve your problem, but I don't have the free time at the moment.

Many thanks for your issue

LeanderGlanda, 2024-01-10:
Hi @Steffen-W,
I finished routing all signals on the PCB. Now i get a different error. 
I tried a few things: 
* selected one PCB track segment
* selected two vias
* selected a whole net (connection tracks from one pin to another)

No matter what I get the following error pop-up
[image]

Solve the issue whenever you have free time for it. No stress from my part.

Many thanks for your help.

Steffen-W, 2024-01-10:
Thank you for your update. That looks more like a problem to me and I will have a look at it soon. After all, I would also like to be able to use Parasitics.

Joshc17m, 2024-02-23:
Hi @Steffen-W ,

I am having same error, but I thought it was because I'm using the version 8 release 3. I did not have any issues when using version 7.10. 

Hope this helps solve the issue as I use this tool often. 

Thanks, 
Josh

Steffen-W, 2024-02-24:
Hi @Joshc17m , it's motivating to hear that you're using the plugin more regularly. Can you please open a new issue. The errors have nothing to do with each other. I have installed kicad 8, but have yet to take a closer look at the changes. I assume that you are using Linux, right?

Joshc17m, 2024-02-24:
Hi @Steffen-W ,

I actually started a new project to make sure it wasn't something else and the plugin worked. I think it might have been how KiCAD migrated that file from 7.10 to 8?? 

I am on Windows 11

Steffen-W, 2024-02-24:
According to the error message, it looks like the file format for the layer stack has changed. It is about reading out how large the distances between the layers are and the copper thickness. Unfortunately I won't be able to fix the error until next weekend. However, I'll see if I can find a quick solution tomorrow.

commit 99c600a4 (2025-03-08): fix #8
closed 2025-03-08 by Steffen-W