Steffen-W/KiCad-Parasitics #9: Error after selecting two objects

findersee, 2024-01-23:
I have selected two pads, but I still get the error to mark only two objects. This error doesn't change even if I select other pads or vias. Can you tell me what I'am doing wrong.

[image]

Platform: Windows
KiCad: 7.0.9
Plugin: 2023.10.27

Steffen-W, 2024-01-24:
Hi @findersee ,

Many thanks for your issue. At first glance, it looks like an unpleasant error. To narrow down the error, it would be very helpful if you could tell me the number of selected elements.

Please change this [plugins/__init__.py#L120](https://github.com/Steffen-W/KiCad-Parasitics/blob/403bd04c0354c9364eaab22801ecf9de29c3f976/plugins/__init__.py#L120)
```python
                message += " Preferably pads or vias."
```
into this
```python
                message += " Preferably pads or vias." + " Selected Items " + str(len(Selected))
```
to get the number of selected elements.

Were you able to use the plugin for any connection or did it not work at any time?

Steffen-W, 2024-01-24:
I have read through your text again and unfortunately realized that the plugin apparently never worked. That should not be the case. Can you try out what you have written. You don't necessarily have to select vias or pads. You can also select two line elements. The starting point of the respective line is then taken as the point (which is usually not obvious).

[image]

findersee, 2024-01-25:
Hi @Steffen-W After updating to KiCad version 7.0.10 and uninstalling plugins. I managed to get it working. Then I tested and found out that kikit plugin somehow breaks this one from working. I don't have use for kikit currently so this one is solved.

closed 2024-01-25 by findersee