Steffen-W/KiCad-Parasitics #12: plugin not starting in kicad 7.03 and 8.03 under linuxubuntu 20.03

Alec1of1, 2024-07-08:
[]https://i.ibb.co/BrK12K0/upload.jpg
[image]
(url)
no module named "numpy"

8 jul 2024

thanks

Steffen-W, 2024-07-08:
Hi @Alec1of1,

Can you try importing the library manually? See picture. You can find this under tools. I actually assumed that numpy is available by default.

[image]

Steffen-W, 2024-07-08:
How did you install kicd? Via APT, Flatpak, manually?

Alec1of1, 2024-07-10:
Hi Steffen Installation was made from lubuntu 20.04 repos by sudo install kicad    it did not contain pip at all
I thought it will fix the problem in lubuntu 22.04 , Iinstall it but no , no pip no numpy .  In 22.04 I use kicad 8 from appimage.

https://sourceforge.net/projects/kicad-appimage/files/  

works flawlessly and small size.

I tried to install python3 numpy with package manager and import in kicad as you show but it cant fined it .

Alec1of1, 2024-07-10:
https://i.ibb.co/2tvTFH7/upload.jpg

Alec1of1, 2024-07-10:
can you please point where need be numpy in file system,  i will tri to put there manually or how to install it properly thru terminal.
If it is not in system out of box it is mines i m not alone.
Thanks for your time.
Alec

Alec1of1, 2024-07-10:
Lubuntu 22.04 folder file stricture 
 /home/x/.local/share/kicad/8.0/plugins
I install python from repos by muon
/home/x/.local/lib/python3.10/site-packages
/home/x/.local/bin

Alec1of1, 2024-07-10:
[](url)https://i.ibb.co/5sdPPRY/upload.jpg

Alec1of1, 2024-07-10:
https://i.ibb.co/TP8mGd9/upload.jpg

Steffen-W, 2024-07-10:
Hi @Alec1of1,

If you have installed it as an AppImage, you would have to install it in the isolated environment. Maybe you are lucky and pip exists.
```python
import subprocess
import sys

def install(package):
    subprocess.check_call([sys.executable, "-m", "pip", "install", package])

install('numpy')
```

[image]

In the other case, I would assume that your system-wide python installation is being used. Try the following.
```
sudo apt update
sudo apt install python3-pip
pip3 --version
pip3 install numpy
```

Alec1of1, 2024-07-10:
Thanks. but not lucky .

[https://i.ibb.co/1fc0pTf/upload.jpg](url)

https://i.ibb.co/XsbPrgG/upload.jpg[](url)

Steffen-W, 2024-07-10:
I can't see the images. In github you can also simply copy and paste the images into the text field.

Alec1of1, 2024-07-10:
[image]

Steffen-W, 2024-07-10:
It will not help in your case but in general an installation with -u should work.
pip install -U numpy

Alec1of1, 2024-07-10:
it is in terminal?

Steffen-W, 2024-07-10:
But it now looks like numpy is installed. Have you already tried whether "import numpy" works in KiPython?

Alec1of1, 2024-07-10:
same error output ,    did not install numpy

Steffen-W, 2024-07-10:
The only thing I would do now is to ask others about discord. If you feel like it, just ask there. You don't need an account or anything. Unfortunately, I can't help you any further. I'm sorry.
https://discord.com/invite/FANuKv8sZn

Alec1of1, 2024-07-10:
Steffen can this tool calculate parasitic of strip line aka transmission line and show it reflection according to speed of light ??

And thanks allot.

Steffen-W, 2024-07-10:
Unfortunately not in the uploaded version. I am in the process of making this possible but unfortunately not yet finished.

Alec1of1, 2024-07-10:
I will suggest to contact kicad and have it in permanent instillation , it will be a beautiful tool. Just in time of design remember energy located in dielectric and strip line is caring only current.

Alec1of1, 2024-07-10:
discort link not function

Steffen-W, 2024-07-10:
Discord: https://www.kicad.org/community/chat/

KiCad is a wonderful program for designing circuits. Unfortunately, there is no native simulation option apart from circuit simulation.

Programs that might help you. There are of course countless ways to reach your goal. Unfortunately, most of them are not beginner-friendly.
QucsStudio (Windows only but free)
CST Studio Suite (expensive)
Advanced Design System, ADS (expensive)
Elmer (I've always wanted to try it out, it's opensoure)

Steffen-W, 2024-07-10:
> I will suggest to contact kicad and have it in permanent instillation , it will be a beautiful tool. Just in time of design remember energy located in dielectric and strip line is caring only current.

I actually have some experience in this area, but unfortunately not enough free time to put it into practice.

Steffen-W, 2025-03-04:
The issue should be solved and will be closed.

closed 2025-03-04 by Steffen-W