Steffen-W/KiCad-Parasitics #16: Does it  only work on top layer ?

cedounet, 2025-04-02:
Hello,

Using Kicad 9.0 on windows and latest KiCad-Parastics (with patch from https://github.com/Steffen-W/KiCad-Parasitics/issues/15) otherwise does not go any far.  (MacOs i have other issues which are porbably install related)

Took a demo project (kit-dev-coldfire-xilinx-...), to make it easier for you to reproduce, only works for traces on top layer.  See below if I take another one:

[image]

Maybe that is not supported?  (Readme mentioned top layer...)

Do let me know of anything I can do to help you,

Thanks a lot,

C/

Steffen-W, 2025-04-05:
Hi, of course it should work. I will hopefully have a look at it tomorrow. I haven't tested that much with KiCad 9 yet. The problem will be that a CU layer stack has never been saved.

Steffen-W, 2025-04-06:
The problem is the changed cu stack from kicad 8 to 9. Here is the following.
Layers [0, 2, 4, 6]  are required but the following layers are parsed:
```python
{0: {'thickness': 0.035, 'name': 'F.Cu', 'abs_height': 0.01}, 1: {'thickness': 0.035, 'name': 'In1.Cu', 'abs_height': 0.525}, 2: {'thickness': 0.035, 'name': 'In2.Cu', 'abs_height': 1.04}, 31: {'thickness': 0.035, 'name': 'B.Cu', 'abs_height': 1.555}}
```

Admittedly the plugin does not work as it should. I have to change a few things.

Steffen-W, 2025-04-06:
Hi @cedounet, the bug should be fixed. please test it on your site, i would also distribute the version via kicad.
[KiCad-Parasitics.zip](https://github.com/user-attachments/files/19623110/KiCad-Parasitics.zip)

[image]

cedounet, 2025-04-09:
Hello,

I did test and forgot to update.  Does indeed work for me.

Thanks a lot.

commit 39e80cab (2025-04-06): fix #16